OpenBCI/Ultracortex #9: .prt file request -- to edit in Solidworks

momonala, 2016-01-26:
Hi, 
We're using openBCI for a senior thesis project to make audiovisual feedback from eeg, ekg, and mmg. We're lovin openBCI, but we're starting to think about headset design. We want to make a few edits to the ultracortex, but the .stl files are not editable in solidworks. Can you post the original CAD files (.prt)? 

Much appreciation, 
Mohit

ThreeForm, 2016-12-16:
Mohit, We spoke via email about this, but I want to follow up on the open issue listing.

While the Ultracortex frame was not created in CAD (so there is no prt file), we made efforts to offer more useful data and files from the development of Ultracortex M4 so that people can modify it or make their own headset.  For CAD related files we offer an IGS files that contains the frame curves, constructions planes, and mounting points (a new M4 feature). Those who contacted us about CAD source (like yourself) are among the "beta" recipients of the templates, so I hope you'll offer us some feedback before we post the template.

momonala, 2016-12-27:
Hey,

Thanks for the update. The previous EEG project I was working on is in a
bit of a hiatus as the team has split up... Won't be much use for beta
testing right now. But I am excited to get my Ganglion next month!
Hopefully some new project ideas will spark up then.

Thanks,
Mohit