OpenBCI/Ultracortex #11: Where are the Mark 4 files?

strfry, 2016-03-18:
The Mark 4 Model is already existing, judging from the pictures on http://shop.openbci.com/products/ultracortex-mark-iv

Why isn't it released yet?

teonbrooks, 2016-03-20:
hi @strfry, the Mark 4 is still in concept-phase. The STL will be released once the final product begins rolling out.

JFD-Mtl, 2016-03-28:
Do you have an expected release date ?

ThreeForm, 2016-05-02:
@strfry The Mark 4 is scheduled to be released in August 2016.

shruna, 2017-10-15:
Where are the bolts which holds electrodes on mark 4, are they same which are used in mark 3

teonbrooks, 2017-10-15:
@conorrussomanno ^

ThreeForm, 2017-10-15:
You can find the M4 bolt, and several variations of it, in the M4 [DevKit hardware section](https://github.com/OpenBCI/Ultracortex/tree/master/Mark_IV/DevKit/M4%20Hardware/Extended%20parts). The M4 bolt is smaller than M3 because of the higher node count in M4 (to stop them overlapping). They weren't in the original "Print your own" file release because the electrode assemblies are injection molded in the M4 kit. That means if you are not buying the electrode kit from OpenBCI you are going to have to source your own springs and electrodes, and you will need to either print your own bolt hardware or buy them on Shapeways here: [M4 headset parts on Shapeways](https://www.shapeways.com/shops/ultrabci)

closed 2022-09-29 by retiutut