OpenBCI/Ultracortex #23: Battery charging?

Abraxas3d, 2016-09-05:
In the instructions about including the rechargeable battery, how is the charger included and connected?

"Embed OpenBCI into the Ultracortex

Connect your ~500 mAh lithium ion rechargeable batter to the back of your 32bit OpenBCI Board. Then fold the battery and its wires neatly behind the board before inserting the board into the BOARD_MOUNT. You can then use one of your #4 Drive Screws to secure the OpenBCI Board to the BOARD_MOUNT. Typically you don't need to use any #4 Drive Screws because the BOARD_COVER locks the OpenBCI Board in place, but in this case it's a good idea because you'll want to see where you're connecting your wires."

onyedikilo, 2017-02-01:
There is no charger, the batteries (which are not included) are AA size batteries. There is only a case to hold the batteries is included. You can also ditch the casing and use a Li-PO rechargeable battery. In both cases you will need to remove the battery and recharge and insert back again.

andreaortuno, 2017-03-13:
We recommend you purchase:

andreaortuno, 2017-03-13:
We recommend you purchase:

(2x) Lithium Ion Rechargeable Battery Pack (500mAh) from [Adafruit](https://www.adafruit.com/product/1578) or [Sparkfun](https://www.sparkfun.com/products/13851) — we recommend buying an extra battery so you can continue working while you recharge!
(1x) A charger for your battery pack ([Adafruit](https://www.adafruit.com/products/1304) or [Sparkfun](https://www.sparkfun.com/products/10217))

The batteries fit in the board case from the Ultracortex, but you will need to charge the batteries separately.

closed 2017-03-13 by andreaortuno