OpenBCI/Ultracortex #30: STL files are not meant to be printed at home

tierralibre, 2017-12-18:
It would be great it the files were designed to be printed in to so not PRO 3d printing machines.

As well the orientation print way too much support. 

Please help!

ThreeForm, 2017-12-18:
The files print on normal size printers with a bed around 22cm. This includes the most popular machines like Prusa i3 (or clones), Ultimaker (or clones), Makerbot (or clones). All production units were made on MakerBot Rep2's and Lulzbot Taz5's.  Prusa clones can cost about $200, and are the most easily sourced printers around (though not as good quality as the original, obviously). Most people having trouble are using incorrect settings.  Make sure your orientation is correct. The flat side should be flat on the bed so the print is stable and doesn't need much support. Total support volume should be 10-20%.  More tips are available on the [OpenBCI Forum](http://openbci.com/forum/index.php?p=/discussion/1174/difficulty-3d-printing-tips-for-printing-from-aaron), or you can just buy an optimized nylon version of the frame on [Shapeways](https://www.shapeways.com/product/WKM73GNJM/m4-ultra-headset-frame?optionId=63113712).

Shootquinn, 2018-06-13:
I did some testing on this, and I have a couple of comments to make.

I got the "Whole Medium MkIV" to print in 14:34 on the Prusa Mk3 without supports. Of course there were issues with this approach, but I'm never afraid to push things. Later I printed one in 41:45 with supports set to 25 degrees in slic3r, which is about the minimum you can do on this before getting some of the failures showing up. 

The OP does have a point, although he isn't presenting it in a collaborative fashion. I do see that during the next redesign if an effort is made to minimize overhangs with asymetrical filleting, slightly changing the angles of the structures between one boss to another, etc., that the printing time can be reduced three fold and with a 15% cost savings based on your claim of support volume above. This would also make it more printable on less capable, cobbled together machines possibly in the 3rd world, support or no support. 

Thanks for keeping an open mind about this, I do love the design but there are some gains to be made as far as the printability goes. 

Note on my 14.5 hour print, it still has more usable bosses than easier versions of the Ultracortex. It could be put in service. 

Quinn

[image]

[image]

[image]

closed 2022-09-29 by retiutut